RedGreenBlue09/WFAv7_Installer #10: BSOD on Spec A

dee-fox, 2023-07-14:
Installation claimed to be successful, logs also show that.
Possibility for client-side errors present: Cyrillic letters used in some path names, Russian version of Windows 10 is used on the deployment machine (making logs unusable).
Device: Nokia Lumia 920 (32 GB, Unlocked, RM-821)

RedGreenBlue09, 2023-07-14:
What kind of blue screen do you see? The one with big sad face in the middle or the one with QR code? Lastly, can you provide the log file?

dee-fox, 2023-07-14:
sad smiley emoticon!
logs are barely useable, but i'll try to grab them right now

RedGreenBlue09, 2023-07-14:
The sad face screen means something is wrong with the BCD file (but I really don't know).

One more question. You used dual-boot or single-boot?

dee-fox, 2023-07-14:
Singleboot. Can't get the log off the computer I used to deploy, will try again (I don't like using windows on desktop)

RedGreenBlue09, 2023-07-14:
Okay. I suggest you to also try the dual-boot method when possible as it doesn't make modifications on the partition layout.

dee-fox, 2023-07-14:
Will do, thanks.

dee-fox, 2023-07-14:
It still doesn't work, unfortunately, the phone still displays a sad face when booting Windows 10 ARM or bootloops if trying to run Windows Phone (8.1). 20k MB size used.

RedGreenBlue09, 2023-07-14:
I will investigate that. Thank you for testing it out.

RedGreenBlue09, 2023-07-22:
Which version of WPinternals did you used to unlock bootloader? Only WPi 2.9 works.

RedGreenBlue09, 2023-07-22:
Can you try again with WPinternals 2.9 and this [latest](https://github.com/RedGreenBlue09/WFAv7_Installer/archive/refs/heads/master.zip) version?

dee-fox, 2023-07-23:
> Which version of WPinternals did you used to unlock bootloader? Only WPi 2.9 works.

hmm, haven't thought about that! i used the one i had lying around, which probably is the latest
i'll see through that again later today

RedGreenBlue09, 2023-08-01:
As of #11, it is confirmed that Spec A installation is working in some cases. Old WPinternals seems to be the culprit.

dee-fox, 2023-08-02:
Well, that's pretty sucky considering I cannot for the love of me figure out why WPi complains about FFU OS-versions

RedGreenBlue09, 2023-08-02:
[image]

If your original FFU version is not there, you will need another donor FFU (yes, both is needed). The donor FFU needs to be anything that it supports. I recommend 950 (XL) FFUs with version 10.0.10586.318.

dee-fox, 2023-08-07:
Alright. I finally got it.
It works, even though that was quite strange behaviour from something that's supposed to just be file manipulation really
Computers really are magic I guess...

closed 2023-08-08 by RedGreenBlue09

RedGreenBlue09, 2023-08-08:
See my wiki for more information about Windows 10 ARM on Lumia.  
https://github.com/RedGreenBlue09/WFAv7_Installer/wiki/Apps-on-Windows-10-ARM