RedGreenBlue09/WFAv7_Installer #11: Lumia 920 stuck at "Nokia" screen with black background

jacob1298, 2023-07-27:
I unlocked the bootloader of my Nokia Lumia 920 and flashed the Engineering SLB3 bootloader using WPInternals 2.9. I then went into Mass Storage Mode and followed the instructions on the readme. After the Installer.cmd file completed successfully, I rebooted my Lumia 920 to find it stuck at the "Nokia" screen. What should I do to fix this?

jacob1298, 2023-07-27:
I got it working after a couple of force reboots and switching in and out of Mass Storage mode.

closed 2023-07-27 by jacob1298

RedGreenBlue09, 2023-07-28:
You got it booted with the latest release or the latest commit?

jacob1298, 2023-07-28:
latest release, but is there a newer windows release?

RedGreenBlue09, 2023-07-28:
There are no newer Windows builds for ARM(v7) that's as good as 15035. For example the build 16299 is available but it only boots into Windows logo. Keep in mind all Windows 10 ARMv7 builds are LEAKED.

RedGreenBlue09, 2023-07-28:
Charging is known to not work. We cannot support all features as this is not supported by Microsoft. We can only do our best with existing Windows Phone drivers.

jacob1298, 2023-07-28:
It charges when it sleeps. Does this use EDK2 or similar? I'd like to look at something like Linux.

RedGreenBlue09, 2023-07-28:
No. Official Windows Phone firmware is already using EDK2-based UEFI and has built in Windows Boot Manager. WPinternals enables the ability to boot other stuff using the stock firmware. WOA (ARM64) on Lumia 950 uses custom EDK2 though.

For Linux, check mainline4lumia. Spec A devices are not supported though.

jacob1298, 2023-07-28:
I'll check it out. I actually have a 950 with Windows 11 on it. It's painfully slow though. The 920's lack of ram makes the browser crash, so hopefully Linux won't be as hard on the hardware.

jacob1298, 2023-07-28:
Just checked, mainline4lumia doesn't support the 920

RedGreenBlue09, 2023-07-28:
> I'll check it out. I actually have a 950 with Windows 11 on it. It's painfully slow though. The 920's lack of ram makes the browser crash, so hopefully Linux won't be as hard on the hardware.

The browser crashed because you don't have the Andreno display driver installed. You need to get into safe mode to update the display driver.

RedGreenBlue09, 2023-07-28:
> Just checked, mainline4lumia doesn't support the 920

The only Linux on Lumia project that supports spec A as I know is android4Lumia. No support for 920 though.

jacob1298, 2023-07-28:
> The browser crashed because you don't have the Andreno display driver installed. You need to get into safe mode to update the display driver.

How do I do that?

RedGreenBlue09, 2023-07-28:
On PC or phone:
```
bcdedit /store <MainOS>:\EFIESP\EFI\Microsoft\Boot\BCD /set {703c511b-98f3-4630-b752-6d177cbfb89c} safeboot minimal
```

Then reboot the phone. After you are in safe mode, open Device Manager and update the basic adapter. You need to select "Pick existing driver on my device" and choose the Andreno driver.

On PC or phone:
```
bcdedit /store <MainOS>:\EFIESP\EFI\Microsoft\Boot\BCD /deletevalue {703c511b-98f3-4630-b752-6d177cbfb89c} safeboot
```

Reboot.

jacob1298, 2023-07-28:
Thanks. I'll try it

jacob1298, 2023-07-28:
I tried it, but got the response "The system cannot find the file specified."

RedGreenBlue09, 2023-07-29:
> I tried it, but got the response "The system cannot find the file specified."

You need to replace <MainOS> with your MainOS drive letter.

jacob1298, 2023-07-29:
I checked before using the command, and the correct display driver is already installed.

RedGreenBlue09, 2023-07-29:
The idk. The browser should work when Andreno driver is installed.

jacob1298, 2023-07-29:
It has only 1gb of ram, and idles at 90% usage

RedGreenBlue09, 2023-08-02:
Hey, you should check the page file settings. It can be disabled or too small. Increase it to a larger size might fix the issue, runs much slower though.

jacob1298, 2023-08-02:
Yup, I forgot to say that I messed with the pagefile, which turned out to
be stuck at 0mb by default. It gave me a warning when I first logged on,
and I thought I fixed it, but didn't. After I actually fixed it, it worked
much better, and the browser worked fine.

RedGreenBlue09, 2023-08-03:
Congrats