RedGreenBlue09/WFAv7_Installer #15: Bootloader unlock error

armX69, 2023-09-08:
Stuck at the first step with my Lumia 640 XL LTE (Phone model: Lumia 640 XL LTE, PackageTitle: RM-1062 VAR EURO 7L CV, Product Code: 059X6M4, manufacturerHardwareModel: RM-1062, Last Firmware: 02177.00000.15221.30001). Trying to unlock the bootloader with the phone ffu, the emergency downloader (should i use the .ede or the .edp file?) and an other model doner ffu with a compatible version, but every time i try to unlock the phone restart one time and the progress bar goes op to 3% but then after the reboot it boots in a red screen with the the same symbol as the flash mode but with a progress bar at 0%. I've left it there working and doing its things for like 15 hours but nothing it remains as i've left it but with the phone in the red screen with the bar and the symbol, and for being able to retry i have to reinstall/repair the fw with windows device recovery tool. Please help me

armX69, 2023-09-08:
Sometimes it doesnt even starts and send me this
[image]

RedGreenBlue09, 2023-09-09:
Hi. The issue you are facing is mostly caused by bad USB connection. Try again with a different cable and port. Maybe change yout PC.

armX69, 2023-09-09:
Thank you after changing pc and cable it worked but now I’m facing 2 new issues 😅: My phone has 8 gb of storage and my data partition can be shrink in only 3,7 gb, is it sufficient for the os?

RedGreenBlue09, 2023-09-09:
Windows Phone will get deleted (Data & MainOS merged & formatted) so yeah not a problem. Dual-booting is not possible.

armX69, 2023-09-10:
Done as you said formatted and merged data and mainos, but in the “configure windows boot manager” section it asks to enter the mainos letter which isn’t anymore existing. What should I do? Thank you in advance for the super support

RedGreenBlue09, 2023-09-10:
It's the merged partition. Btw the guide in the wiki is not for single-booting. Use the installer, it is fully automatic.

armX69, 2023-09-10:
Ok windows installed and I’ll remove the issue. Thank you 1000 for the support and for the lightspeed replies

RedGreenBlue09, 2023-09-10:
You're welcome. Remember to check the wiki for usage guide.

RedGreenBlue09, 2023-09-10:
Close as resolved.

closed 2023-09-10 by RedGreenBlue09