RedGreenBlue09/WFAv7_Installer #19: Driver Downloader

ankopriv, 2023-09-22:
Driver downloader does not download drivers for Lumia1520AT&T. When choosing this option only txt file "1520attupsidedown" is being downloaded the components folder is missing. The option for Lumia1520 does download the drivers correctly, however I dont know if this can be used in AT&T version.

RedGreenBlue09, 2023-09-22:
This is a bug. I'm fixing it.

RedGreenBlue09, 2023-09-22:
[Driver Downloader.zip](https://github.com/RedGreenBlue09/WFAv7_Installer/files/12702109/Driver.Downloader.zip)
Replace the existing file with this one. This will be released later.

RedGreenBlue09, 2023-09-22:
> however I dont know if this can be used in AT&T version.
No that doesn't work. Some drivers (even same soc) are device specific which will crash with the wrong phone.

RedGreenBlue09, 2023-09-22:
New version with this fix released.

RedGreenBlue09, 2023-09-23:
Is this issue resolved?

ankopriv, 2023-09-25:
Yes this is resolved the downlaoder works, thank you!

closed 2023-09-25 by ankopriv