RedGreenBlue09/WFAv7_Installer #22: Having an issue where my Lumia 520 freezes on bootup.

CitroTheWolf, 2023-11-26:
It boots to the boot manager, but freezes when the spinning dots loads up. Anything I can do?

CitroTheWolf, 2023-11-28:
Okay! Ill try! Thanks!

CitroTheWolf, 2023-11-28:
Im stuck, am i supposed to up click on the windows 10 (debugger) thing?

CitroTheWolf, 2023-11-28:
************* Preparing the environment for Debugger Extensions Gallery repositories **************
   ExtensionRepository : Implicit
   UseExperimentalFeatureForNugetShare : true
   AllowNugetExeUpdate : true
   AllowNugetMSCredentialProviderInstall : true
   AllowParallelInitializationOfLocalRepositories : true

   -- Configuring repositories
      ----> Repository : LocalInstalled, Enabled: true
      ----> Repository : UserExtensions, Enabled: true

>>>>>>>>>>>>> Preparing the environment for Debugger Extensions Gallery repositories completed, duration 0.000 seconds

************* Waiting for Debugger Extensions Gallery to Initialize **************

>>>>>>>>>>>>> Waiting for Debugger Extensions Gallery to Initialize completed, duration 0.031 seconds
   ----> Repository : UserExtensions, Enabled: true, Packages count: 0
   ----> Repository : LocalInstalled, Enabled: true, Packages count: 36

Microsoft (R) Windows Debugger Version 10.0.25921.1001 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Using USB for debugging
Waiting to reconnect...
USB: Write opened
Connected to Windows 10 15035 ARM (NT) Thumb-2 target at (Tue Nov 28 17:45:37.678 2023 (UTC - 6:00)), ptr64 FALSE
Kernel Debugger connection established.
Symbol search path is: srv*
Executable search path is: 
*************************************************************************
***                                                                   ***
***                                                                   ***
***    Either you specified an unqualified symbol, or your debugger   ***
***    doesn't have full symbol information.  Unqualified symbol      ***
***    resolution is turned off by default. Please either specify a   ***
***    fully qualified symbol module!symbolname, or enable resolution ***
***    of unqualified symbols by typing ".symopt- 100". Note that     ***
***    enabling unqualified symbol resolution with network symbol     ***
***    server shares in the symbol path may cause the debugger to     ***
***    appear to hang for long periods of time when an incorrect      ***
***    symbol name is typed or the network symbol server is down.     ***
***                                                                   ***
***    For some commands to work properly, your symbol path           ***
***    must point to .pdb files that have full type information.      ***
***                                                                   ***
***    Certain .pdb files (such as the public OS symbols) do not      ***
***    contain the required information.  Contact the group that      ***
***    provided you with these symbols if you need this command to    ***
***    work.                                                          ***
***                                                                   ***
***    Type referenced: nt!_MMPTE_TRANSITION                          ***
***                                                                   ***
*************************************************************************
Windows 10 Kernel Version 15035 MP (1 procs) Free ARM (NT) Thumb-2
Edition build lab: 15035.0.armfre.rs2_release.170209-1535
Kernel base = 0x82857000 PsLoadedModuleList = 0x82a64438
System Uptime: 0 days 0:00:00.362
*************************************************************************
***                                                                   ***
***                                                                   ***
***    Either you specified an unqualified symbol, or your debugger   ***
***    doesn't have full symbol information.  Unqualified symbol      ***
***    resolution is turned off by default. Please either specify a   ***
***    fully qualified symbol module!symbolname, or enable resolution ***
***    of unqualified symbols by typing ".symopt- 100". Note that     ***
***    enabling unqualified symbol resolution with network symbol     ***
***    server shares in the symbol path may cause the debugger to     ***
***    appear to hang for long periods of time when an incorrect      ***
***    symbol name is typed or the network symbol server is down.     ***
***                                                                   ***
[... 211 lines cut ...]
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
[... cut ...]
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.

CitroTheWolf, 2023-11-28:
There! i got it, i think.

CitroTheWolf, 2023-11-28:
Also how do i get access to the storage?

CitroTheWolf, 2023-11-28:
Im a bit of a newbie to this type of modification, I'm just starting to get my IT credits for college.

CitroTheWolf, 2023-11-28:
It seems to be more of a conflict with the oobe?

CitroTheWolf, 2023-11-29:
it crashed? WER/CrashExp:363: ERROR Failed to create the process "C:\Windows\system32\WerFault.exe", cmdline "C:\Windows\system32\WerFault.exe -u -p 3656 -s 2208", creationflags 525312, HRESULT 800705AF

CitroTheWolf, 2023-11-29:
Page file too small. Yup.

RedGreenBlue09, 2023-11-29:
> Also how do i get access to the storage?

I think low RAM is the issue. Try increasing the page file size.

You can access phone storage via Mass Storage Mode.

CitroTheWolf, 2023-11-29:
> > Also how do i get access to the storage?
> 
> 
> 
> I think low RAM is the issue. Try increasing the page file size.
> 
> 
> 
> You can access phone storage via Mass Storage Mode.

So how do i get into mass storage mode? its usually done by going into WPinternals, but it doesnt show up on wpinternals.

RedGreenBlue09, 2023-11-29:
Sorry for the slow reply. I'm quite busy.

CitroTheWolf, 2023-11-29:
> > So how do i get into mass storage mode? its usually done by going into WPinternals, but it doesnt show up on wpinternals.
> 
> 
> 
> In WPinternals: Click `interrupt boot process`, then reboot the phone.
> 
> 
> 
> > If its possible i can put the pagefile on a SD Card, the storage space is incredibly small
> 
> 
> 
> A SD Card as it's marked as a removable storage, but you cannot put the page file on a removable storage AFAIK. You don't know the drive letter of the SD Card anyway.
> 
> 
> 
> There is a thing called [ReadyBoost](https://en.wikipedia.org/wiki/ReadyBoost) for removable storage which can help reducing RAM usage, you can enable that after OOBE.
> 
> 
> 
> > Ive ran windows 10 on 512MB fine, so this is genuinely surprising for me lol
> 
> 
> 
> It worked on 512 MB thanks to the page file. Windows 10 ARM somehow fails to create the page file automatically, so you need to add it manually.

Okay! and I don't mind! I appreciate your help greatly sir!

RedGreenBlue09, 2023-11-30:
I mentioned that here:

> [mounting it's registry](https://smallvoid.com/article/winnt-offline-registry-edit.html) and [enable the page file](https://www.itprotoday.com/cloud-computing/how-can-i-remotely-change-size-pagefile).

CitroTheWolf, 2023-11-30:
> I mentioned that here:
> 
> 
> 
> > [mounting it's registry](https://smallvoid.com/article/winnt-offline-registry-edit.html) and [enable the page file](https://www.itprotoday.com/cloud-computing/how-can-i-remotely-change-size-pagefile).

- [ ] I did it lol, im just gonna wipe win10 and put WinPE or sum on it. im tryna unlock it rn

CitroTheWolf, 2023-12-07:
i cant unlock my bootloader

RedGreenBlue09, 2024-01-10:
The best I can predict about this issue is that we have insufficient RAM because the page file for some reason fails to create automatically. Unfortunately, I don't own the 520 so I can't test it myself. I'll consider making a page file fix instead.

stx3plus1, 2024-01-13:
oh damn, I was also getting this issue but THERE IS A FIX. disabling testsigning makes it work!? but then touch and stuff is screwed up. it seems to be a **driver** issue somehow and im not sure how this could be fixed personally. windows boots to the oobe on my 520 when testsigning is off

stx3plus1, 2024-01-13:
im gonna flash my 520 (it has rt on it) and try fixing this issue since it happened twice in a row and i sorta gave up :P

RedGreenBlue09, 2024-01-13:
I think if you disable testsigning, less drivers load hence less RAM used so it don't freeze.

stx3plus1, 2024-01-13:
would make sense, but its rather odd and CurrentControlSet doesnt seem to yet exist when windows freezes, maybe letting it get to oobe will make this work? controlset001 does exist and applying the mod there it still shows the frozen dots (windbg is still responding with the hardware buttons)

stx3plus1, 2024-01-13:
disabling the registry settings to make windows boot to the regular logonui freezes on a total black screen with the backlight illuminated. might try reading the logs after clearing windbg and rebooting - the phone is still on and responding but windows at least gets _somewhere_ now it seems. no user account to log onto though if logonui does work, but i expect then replacing the cmdline would let us use osk to make a user or just make a batch file on pc to do it

stx3plus1, 2024-01-13:
[custommadewindbglog.txt](https://github.com/RedGreenBlue09/WFAv7_Installer/files/13929430/custommadewindbglog.txt)
windows gets this far according to windbg (might have missed ntoskrnl doing some stuff tho, time was tight between the usb write starting and NT starting)

stx3plus1, 2024-01-15:
SOLVED IT! my pagefile was too small, set it to minimum 1024 max 2048 and win10 booted like a charm. thank you!!

stx3plus1, 2024-01-15:
15035 isnt aware of the memory issue and needs to be forced to use the larger pagefile or the display just wont work i guess? i wonder if the original post can get this working too, and if so it could be implemented for 512MB devices as a quick mod with the reg command line tool?

RedGreenBlue09, 2024-01-16:
Congrats! But how did you change the pagefile size? Using regedit?

stx3plus1, 2024-01-16:
yep, using regedit and forcing a minimum size of 1024. the phone also has this really strange thing where it wont boot with testsigning on or off and when i disable "nobootuxprogress" theres an infinite spinning dots animation from the bootscreen. could this be registry related (i have tampered with the setup keys)

RedGreenBlue09, 2024-01-17:
testsigning should be always on. You can screw up the installation if you disables it.

stx3plus1, 2024-01-20:
the exact same thing is happening with fresh windows 10 lol looking for a fix

CitroTheWolf, 2024-01-21:
> the exact same thing is happening with fresh windows 10 lol looking for a fix

im surprised people are having the same issue. thank you guys for the help, also how do i like reunlock the bootloader lol

RedGreenBlue09, 2024-01-21:
> > the exact same thing is happening with fresh windows 10 lol looking for a fix
> 
> im surprised people are having the same issue. thank you guys for the help, also how do i like reunlock the bootloader lol

Maybe we can help you with that in the telegram group, but remember, support is not guaranteed.

RedGreenBlue09, 2024-03-31:
Hi everybody, the page file fix is added in 842200409bf2eb575328a63dfb3fc07da64a4630. Download the master branch to try it out.  
https://github.com/RedGreenBlue09/WFAv7_Installer/archive/refs/heads/master.zip

Eeveelution, 2024-04-02:
Even with the latest commit I'm getting the same errors as them with it not being able to get the OOBE Done state. The first time it happens it fails and then the loading spinner appears, after the reboot it gets the same error but doesn't get the loading spinner or anything, it just stays there with the Nokia logo...

Here's the full log, the very bottom is the attempt that just freezes:

```

************* Preparing the environment for Debugger Extensions Gallery repositories **************
   ExtensionRepository : Implicit
   UseExperimentalFeatureForNugetShare : true
   AllowNugetExeUpdate : true
   NonInteractiveNuget : true
   AllowNugetMSCredentialProviderInstall : true
   AllowParallelInitializationOfLocalRepositories : true

   EnableRedirectToV8JsProvider : false

   -- Configuring repositories
      ----> Repository : LocalInstalled, Enabled: true
      ----> Repository : UserExtensions, Enabled: true

>>>>>>>>>>>>> Preparing the environment for Debugger Extensions Gallery repositories completed, duration 0.000 seconds

************* Waiting for Debugger Extensions Gallery to Initialize **************

>>>>>>>>>>>>> Waiting for Debugger Extensions Gallery to Initialize completed, duration 0.031 seconds
   ----> Repository : UserExtensions, Enabled: true, Packages count: 0
   ----> Repository : LocalInstalled, Enabled: true, Packages count: 41

Microsoft (R) Windows Debugger Version 10.0.27553.1004 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Using USB for debugging
Waiting to reconnect...
USB: Write opened
Connected to Windows 10 15035 ARM (NT) Thumb-2 target at (Tue Apr  2 15:25:08.337 2024 (UTC + 2:00)), ptr64 FALSE
Kernel Debugger connection established.  (Initial Breakpoint requested)
Symbol search path is: srv*
Executable search path is: 
*************************************************************************
***                                                                   ***
***                                                                   ***
***    Either you specified an unqualified symbol, or your debugger   ***
***    doesn't have full symbol information.  Unqualified symbol      ***
***    resolution is turned off by default. Please either specify a   ***
***    fully qualified symbol module!symbolname, or enable resolution ***
***    of unqualified symbols by typing ".symopt- 100". Note that     ***
***    enabling unqualified symbol resolution with network symbol     ***
***    server shares in the symbol path may cause the debugger to     ***
***    appear to hang for long periods of time when an incorrect      ***
***    symbol name is typed or the network symbol server is down.     ***
***                                                                   ***
***    For some commands to work properly, your symbol path           ***
***    must point to .pdb files that have full type information.      ***
***                                                                   ***
***    Certain .pdb files (such as the public OS symbols) do not      ***
***    contain the required information.  Contact the group that      ***
***    provided you with these symbols if you need this command to    ***
***    work.                                                          ***
***                                                                   ***
***    Type referenced: nt!_MMPTE_TRANSITION                          ***
***                                                                   ***
*************************************************************************
Windows 10 Kernel Version 15035 MP (1 procs) Free ARM (NT) Thumb-2
Edition build lab: 15035.0.armfre.rs2_release.170209-1535
Kernel base = 0x8285f000 PsLoadedModuleList = 0x82a6c438
System Uptime: 0 days 0:00:00.250
*************************************************************************
***                                                                   ***
***                                                                   ***
***    Either you specified an unqualified symbol, or your debugger   ***
[... 878 lines cut ...]
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
[... cut ...]
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.

```

Eeveelution, 2024-04-03:
Hmm, sad.. Yeah I actually did experiment with adding the registry keys it was complaining about but it didn't make a difference. I let it boot for another hour but it didn't do anything

stx3plus1, 2024-04-06:
i will try the fix and changing some other aspects of the system to just try get it booting, however i might need to do a DPP transplant from another lumia as mine is running TWRP... i could probably try emulating the situation on some other lumias too

RedGreenBlue09, 2024-04-06:
DPP transplant? I don't think that will work properly as it is supposed to be unique per phone.

stx3plus1, 2024-09-21:
I have in fact done some partition scheme war crimes on WP and had it work, uniqueness doesn't matter for this old of a device...

I am working on obtaining another 520 so I can debug on that device further, I switched my first one to Android and the backup of the system partitions corrupted.. as I previously reported the system without any drivers injected boots to "Get going fast" without issue, however obviously input is non-functional. I will take a skim through the drivers injected and what exactly the hardware is trying and visibly failing to do as the system is also clearly running (hardware button interrupts) after the display freezes up.

stx3plus1, 2024-09-27:
+1 Lumia 520. Horrors of debugging after school start today (its Friday however, yay!)

stx3plus1, 2024-09-27:
BREAKTHROUGH: First attempt has shown windeploy messages and exited the stage successfully. This issue might be resolved after all and hasn't been tested.

stx3plus1, 2024-09-27:
[image]
This is new.

stx3plus1, 2024-09-27:
windeploy ran correctly. ONCE. Then blackscreened with no clear indication of what is happening. I will set the pagefile to a higher amount before attempting other debug methods for this... hard reset and back to the good old freezing dots :D

stx3plus1, 2024-09-27:
[image]
[image]
This is new too. Obviously a driver. Doesn't exist on my host.

RedGreenBlue09, 2024-09-28:
If WinDBG didn't help, just try with the minimal number of drivers and add more each time.

stx3plus1, 2024-09-28:
Tried to remove the driver that caused the bugcheck above just in case, as it isn't hard to reinstall, which was installed as oem23.inf and simplehw.inf in the driver packages, but now this is happening... This is a Windows Phone pulled driver however.
```
WARNING: nt_82449000 overlaps nt
WARNING: nt_82449000 overlaps nt_8280d000
WARNING: nt_82449000 overlaps hal
WARNING: hal overlaps nt
WARNING: kdcom overlaps nt_8280d000
BGFX Display Ready Time (ms): 22706
BGFX Secondary Logo Bitmap Display Time GOAL (ms): 27706
EXWORKER: worker exit with system affinity set, worker routine 8CB05629, parameter 8BB82B74, item 8BB82B74
KDTARGET: Refreshing KD connection
BGFX Secondary Logo Bitmap Display Time (ms): 27725
WARNING: netbt overlaps qcacdb8930_92740000
WARNING: afd overlaps qcacdb8930_92740000
WARNING: afd overlaps qcchipinfo8930_927d0000
WARNING: afd overlaps hwnled_927e0000
WARNING: vwififlt overlaps qcmbrg8930_92410000
WARNING: pacer overlaps qcremoteat8930_92430000
WARNING: pacer overlaps nokiadataloggerdriver_92440000
WARNING: rdbss overlaps CAD_92480000
WARNING: rdbss overlaps NdisVirtualBus_92490000
WARNING: rdbss overlaps swenum_924a0000
WARNING: rdbss overlaps ks_924b0000
WARNING: csc overlaps ks_924b0000
WARNING: csc overlaps rdpbus_92500000
WARNING: csc overlaps qcpmicapps8930_92510000
WARNING: qcadsp8930 overlaps HIDCLASS_926c0000
WARNING: fastfat overlaps HIDCLASS_926c0000
WARNING: fastfat overlaps filecrypt_926e0000
WARNING: fastfat overlaps tbs_926f0000
WARNING: fastfat overlaps Null_92700000
WARNING: fastfat overlaps Beep_92710000
DMUX::BamMuxRegGetNumBuffers:2319: Number of buffers = 256
ST: Welcome To Synaptics
WARNING: NrsDriver_a1930000 overlaps kbdclass
ST: OpenIOTarget Entry
ST: OpenIOTarget Exit
ST: Starting bring up sequence for the controller
ST: Setting reset gpio pin to low
ST: Waiting...
ST: Setting reset gpio pin to high
ST: Waiting...
ST: Done
ST: Error retrieving registry configuration - STATUS:C0000034
ST: Error reading registry config, using defaults! - STATUS:C0000034
ST: Driver tried enable backlight but HwnReady = false
ST: Discovered function $34
ST: Discovered function $1
ST: Discovered function $11
ST: Discovered function $54
ST: Discovered function $31
ST: Discovered function $1a
ST: Discovered function $55
ST: Discovered function $51
ST: Discovered 8 RMI functions total
ST: founded 1 interupts for f11. setting mast to 4
ST: founded 1 interupts for f1a. setting mast to 10
ST: Max Fingers Count. Value is 5
ST: Error finding idle notification request in IdleQueue:0x5FB96660 - STATUS:8000001A
ST: Created hidReportDescBuffer on A04AC640
ST: Set X=480 in A04AC666
ST: Set Y=800 in A04AC67B
ST: Set X=480 in A04AC6A9
ST: Set Y=800 in A04AC6BE
ST: Set X=480 and Y=800 in hidReportDescriptor
ST: Unexpected -- no interrupt status bit set
Capacity:1331, FullChargedCapacity:1350, Voltage:4079, Rate:0
WARNING: qcsitarcodec8930 overlaps qcmbhcgpio8930
WARNING: qcsitarcodec8930 overlaps qc8x60aud
LumiaButtonsGPIO: LumiaButtonsGPIOProbeResources Entry
[... 94 lines cut ...]
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
WARNING: cldflt overlaps qccamavs8930
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
WARNING: mrxsmb20 overlaps NokiaEnergyDriver
WARNING: mrxsmb20 overlaps BATTC
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OOBE doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\OEM\Nokia\Multivariant\Color doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Shell\OEM\Wallpaper doesn't exist !!!!
NRSDRIVER::NRSDebugPrintState (ERR): Key \Registry\Machine\Software\Microsoft\Windows\CurrentVersion\Control Panel\Theme doesn't exist !!!!
NRSDRIVER::NRSSetDeviceColor (ERR): Getting OOBE done state failed.
WARNING: srvnet_a1bc0000 overlaps srv2
```
Overlap hell?

RedGreenBlue09, 2024-09-28:
Well, I don't think the WinDBG log is going to help dummies like us. I would rather try only with the core drivers that are just enough for it to boot up, then add more (half each time) until it breaks.

Which are the core drivers? Idk, but I think are these:
- `QC8930\PLATFORM.SOC_QC8930.BASE\Source\Drivers\HALs`
- `QC8930\PLATFORM.SOC_QC8930.BASE\Source\Drivers\SOC`
- `ANYSOC\Support`

stx3plus1, 2024-09-28:
Will try this now. Was out for an hour. With only the core in early 2024 it did actually boot _just fine_ but I guess it is time to find the misbehaving driver.

RedGreenBlue09, 2024-09-28:
You can use the divide and conquer strat: Divide the remaining drivers by 2, then if one half have the problem, continue to divide that half by 2 and so on. Should be quick.

stx3plus1, 2024-09-29:
Nice idea. I'll remove oem*.inf. I main Linux so progress is pretty slow since my system is often in usage as a server, but I'll get started

RedGreenBlue09, 2024-09-29:
I suggest a clean install every time if that doesnt work.

stx3plus1, 2024-09-29:
All the SIGNED drivers I have actually tested to work... so many a Lumia WOA driver issue. I haven't tried specific orders to install but without the OEM drivers the system just bugchecks due to "missing symbols" and whatever.

RedGreenBlue09, 2024-09-29:
If reg method works, then probably stick to it for now. I think the issue is in a driver that reg method doesn't enable.

stx3plus1, 2024-09-29:
I did also see that RT works just fine for some people, so I'll check with the reg method.

Eeveelution, 2024-09-29:
I was able to install Windows RT on my Lumia 520 after some difficulties (mainly some driver issues and the fact that Windows OOBE needs the install to go flawlessly). I had to disable the Qualcomm Encode and Decode drivers for it to not crash immediately, and use a different GPIO driver if i recall correctly

stx3plus1, 2024-09-29:
Woah. Yeah, just got it booting too trying my previous method (enabled built-in admin using a BCPE and a preset script), but here touch doesn't work, it's unsigned I believe. Also, the GPIO driver here did cause a bugcheck (but I didn't get the crashdump), leading me to believe something's up with it as a whole and affecting these issues. I'll try removing the encode and decode drivers then, if I can find their IDs.

stx3plus1, 2024-09-29:
[image]
Damn. This means it's a driver that can be only loaded in test mode, and that rules out the signed drivers as being faulty. I'll check the packages for the device "fame" in particular when I get home, but I'm still happy to see progress.

Eeveelution, 2024-09-29:
[msgpiowin32(1).zip](https://github.com/user-attachments/files/17180288/msgpiowin32.1.zip)

Not sure how much it helps but this is the GPIO driver that works under RT

Eeveelution, 2024-09-29:
regarding the encode/decode driver, I think it was something like qrencode-something.dll, i dont remember the exact name. just renaming it to something else (like putting an underscore behind .dll) made it no longer crash

RedGreenBlue09, 2024-09-29:
Thank you guys for the debuggings. Hope the problematic driver will be pointed out so we can add a workaround in the driver repo.

stx3plus1, 2024-09-29:
Thanks for the non-faulty driver at least under RT. 30 minutes - 1 hour and I'll start on that, and check the digital signature of the original in case it's invalid.

stx3plus1, 2024-09-29:
Can't find anything about a driver with "Encoder" or "Decoder" in the metadata. All I know is it requires test mode to be enabled, and is only present in Windows 10 (Lumia WoA driver potentially). Synaptics appears to be OK from my testing, but don't say this is "confirmed". I'm probably going to restore the phone and clean install 15035 before proceeding, or use the RT installer to dump the drivers and inspect what's going on between RT and 10.

[10 more replies]

closed 2025-10-25 by RedGreenBlue09