RedGreenBlue09/WFAv7_Installer #24: lumia 1020

baotg080599, 2023-12-23:
my lumia 1020 global is blue screen when choose boot window 10

baotg080599, 2023-12-23:
#### INSTALLATION WAS STARTED AT 18:51:36.71 #### 
========================================================= 
## Device is C  ## 
## MainOS is E: ## 
 
## MainOS DriveLetter is E: ## 
## DPP PN is 1 ## 
## EFIESP PN is 21 ## 
## Data PN is 28 ## 
## Dualboot is Y ## 
## Win10SizeMB is 16000 ## 
The type of the file system is NTFS.

Stage 1: Examining basic file system structure ...
Progress: 0 of 15616 done; Stage:  0%; Total:  0%; ETA:   0:00:46    
Progress: 4353 of 15616 done; Stage: 27%; Total:  9%; ETA:   0:00:43 .  
Progress: 10105 of 15616 done; Stage: 64%; Total: 21%; ETA:   0:00:38 .. 
Progress: 15616 of 15616 done; Stage: 100%; Total: 32%; ETA:   0:00:03 ...
                                                                                       
                                                                                       
  15616 file records processed.                                                        
File verification completed.
 Phase duration (File record verification): 1.34 seconds.
Progress: 1 of 1 done; Stage: 100%; Total: 21%; ETA:   0:00:03    
                                                                                       
                                                                                       
  1 large file records processed.                                   
 Phase duration (Orphan file record recovery): 0.05 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 21%; ETA:   0:00:03 .  
                                                                                       
                                                                                       
  0 bad file records processed.                                     
 Phase duration (Bad file record checking): 0.04 milliseconds.

Stage 2: Examining file name linkage ...
Progress: 10065 of 28050 done; Stage: 35%; Total: 35%; ETA:   0:00:03 .. 
Progress: 1 of 1 done; Stage: 100%; Total: 43%; ETA:   0:00:03 ...
                                                                                       
                                                                                       
  1 reparse records processed.                                      
Progress: 17547 of 28050 done; Stage: 62%; Total: 47%; ETA:   0:00:03    
Progress: 18725 of 28050 done; Stage: 66%; Total: 50%; ETA:   0:00:03 .  
Progress: 19035 of 28050 done; Stage: 67%; Total: 52%; ETA:   0:00:03 .. 
Progress: 19409 of 28050 done; Stage: 69%; Total: 53%; ETA:   0:00:03 ...
Progress: 19570 of 28050 done; Stage: 69%; Total: 54%; ETA:   0:00:03    
Progress: 20251 of 28050 done; Stage: 72%; Total: 56%; ETA:   0:00:03 .  
Progress: 20838 of 28050 done; Stage: 74%; Total: 58%; ETA:   0:00:03 .. 
Progress: 21619 of 28050 done; Stage: 77%; Total: 60%; ETA:   0:00:03 ...
Progress: 28050 of 28050 done; Stage: 100%; Total: 61%; ETA:   0:00:03    
                                                                                       
                                                                                       
  28050 index entries processed.                                                       
Index verification completed.
 Phase duration (Index verification): 4.21 seconds.
Progress: 0 of 0 done; Stage: 99%; Total: 61%; ETA:   0:00:03 .  
                                                                                       
                                                                                       
  0 unindexed files scanned.                                        
 Phase duration (Orphan reconnection): 1.85 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 61%; ETA:   0:00:03 .. 
                                                                                       
                                                                                       
  0 unindexed files recovered to lost and found.                    
 Phase duration (Orphan recovery to lost and found): 1.66 milliseconds.
Progress: 1 of 1 done; Stage: 100%; Total: 61%; ETA:   0:00:03 ...
                                                                                       
                                                                                       
  1 reparse records processed.                                      
 Phase duration (Reparse point and Object ID verification): 0.78 milliseconds.

[... 577 lines cut ...]
The operation completed successfully.
========================================================= 
The operation completed successfully.

Microsoft DiskPart version 10.0.22621.1

Copyright (C) Microsoft Corporation.
On computer: DELL

Disk 1 is now the selected disk.

Partition 21 is now the selected partition.

DiskPart successfully set the partition ID.
        1 file(s) copied.

Microsoft DiskPart version 10.0.22621.1

Copyright (C) Microsoft Corporation.
On computer: DELL

DiskPart successfully selected the virtual disk file.

DiskPart successfully detached the virtual disk file.
#### INSTALLATION COMPLETED SUCCESSFULLY ####

RedGreenBlue09, 2024-01-02:
You got QR code or sad face?

baotg080599, 2024-01-03:
> You got QR code or sad face?

I got sad face

RedGreenBlue09, 2024-01-03:
Did you unlocked with WPinternals 2.8 or 2.9? For spec A devices, only 2.9 works. Sorry for the slow response.

baotg080599, 2024-01-03:
I unlocked with WPinternals 2.8, thanks you very much, i will unlock it with WPinternals 2.9