RedGreenBlue09/WFAv7_Installer #25: Lumia 640 (dempsey)

jhay06, 2024-01-02:
Can you add the lumia 640 (microsoft-dempsey) rm_1077 ? 
i have unlocked bootloader , but theres no much i can do.

RedGreenBlue09, 2024-01-02:
640 has no INF driver pack (I'm not the one who make such pack). Furthermore, even with reg method (my installer doesn't supports), it still doesn't have working touch out of box and it requires some manual work to bring it up.

jhay06, 2024-07-16:
hope someone make it work :) thanks

RedGreenBlue09, 2024-07-16:
We are trying with our downstream driver pack, but we are unable to find someone with the 640 who are willing to debug it.

NimiPvPGamer, 2024-07-20:
@RedGreenBlue09 i have a 640 that i could maybe do it
this also exists: https://windowsdev.gitbook.io/lumia-woa32/wstep

RedGreenBlue09, 2024-07-23:
Hi @NimiPvPGamer. Thank you for your attention. You may join the Telegram group and I will provide instructions there.

Sorry for the slow response.

jhayar0719, 2024-09-14:
> @RedGreenBlue09 i have a 640 that i could maybe do it this also exists: https://windowsdev.gitbook.io/lumia-woa32/wstep

hope that you can make it

RedGreenBlue09, 2024-11-13:
It is now supported! Try the latest version.

closed 2024-11-13 by RedGreenBlue09

jhay06, 2024-12-02:
wow its been a while then i saw it is now supported will try it thanka