RedGreenBlue09/WFAv7_Installer #30: After instaling win 10 arm on my lumia 930 with 14393.1066 bluescreens

Intelinsidecom, 2024-01-07:


RedGreenBlue09, 2024-01-07:
The blue screen has QR code or sad face? Can you send me logs?

Intelinsidecom, 2024-01-07:
Sad face

Intelinsidecom, 2024-01-07:
#### INSTALLATION WAS STARTED AT 14:12:06.56 #### 
========================================================= 
## Device is 1  ## 
## MainOS is L: ## 
 
## MainOS DriveLetter is L: ## 
## DPP PN is 1 ## 
## EFIESP PN is 25 ## 
## Data PN is 28 ## 
## Dualboot is y ## 
## Win10SizeMB is 16000 ## 
The type of the file system is NTFS.
Volume label is Data.

Stage 1: Examining basic file system structure ...
Progress: 0 of 3584 done; Stage:  0%; Total:  0%; ETA:   0:00:12    
Progress: 3584 of 3584 done; Stage: 100%; Total: 28%; ETA:   0:00:09 .  
                                                                                       
                                                                                       
  3584 file records processed.                                                        

File verification completed.
 Phase duration (File record verification): 260.94 milliseconds.
Progress: 1 of 1 done; Stage: 100%; Total: 13%; ETA:   0:00:10 .. 
                                                                                       
                                                                                       
  1 large file records processed.                                   

 Phase duration (Orphan file record recovery): 0.00 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 13%; ETA:   0:00:11 ...
                                                                                       
                                                                                       
  0 bad file records processed.                                     

 Phase duration (Bad file record checking): 0.12 milliseconds.

Stage 2: Examining file name linkage ...
Progress: 3587 of 9412 done; Stage: 38%; Total: 29%; ETA:   0:00:19    
Progress: 1 of 1 done; Stage: 100%; Total: 29%; ETA:   0:00:19 .  
                                                                                       
                                                                                       
  1 reparse records processed.                                      

Progress: 9412 of 9412 done; Stage: 100%; Total: 46%; ETA:   0:00:14 .. 
                                                                                       
                                                                                       
  9412 index entries processed.                                                       

Index verification completed.
 Phase duration (Index verification): 568.36 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 46%; ETA:   0:00:14 ...
                                                                                       
                                                                                       
  0 unindexed files scanned.                                        

 Phase duration (Orphan reconnection): 2.47 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 46%; ETA:   0:00:14    
                                                                                       
                                                                                       
  0 unindexed files recovered to lost and found.                    

 Phase duration (Orphan recovery to lost and found): 2.68 milliseconds.
Progress: 1 of 1 done; Stage: 100%; Total: 46%; ETA:   0:00:14 .  
                                                                                       
                                                                                       
  1 reparse records processed.                                      

 Phase duration (Reparse point and Object ID verification): 0.64 milliseconds.

Stage 3: Examining security descriptors ...
[... 230 lines cut ...]
[========================   42.0%                          ] 

[========================   43.0%                          ] 

[=========================  44.0%                          ] 

[========================== 45.0%                          ] 

[========================== 46.0%                          ] 

[===========================47.0%                          ] 

[===========================48.0%                          ] 

[===========================49.0%                          ] 

[===========================50.0%                          ] 

[===========================51.0%                          ] 

[===========================52.0%                          ] 

[===========================53.0%                          ] 

[===========================54.0%                          ]

RedGreenBlue09, 2024-01-07:
Pls drag the file instead of the text.

closed 2024-01-07 by Intelinsidecom