RedGreenBlue09/WFAv7_Installer #34: I get sad face Lumia 930

Intelinsidecom, 2024-01-10:
I unlocked bootloader using latest wpi internals, latest wfav7 installer on win 10 mobile 14393.x, help me!!!

RedGreenBlue09, 2024-01-10:
I need the log file

Intelinsidecom, 2024-01-10:
[09-Jan-24.log](https://github.com/RedGreenBlue09/WFAv7_Installer/files/13886886/09-Jan-24.log)

RedGreenBlue09, 2024-01-10:
The log looks perfectly fine. Therefore, I have no idea why that happens. Maybe I'll have to test it out myself again with build 14393.

RedGreenBlue09, 2024-01-10:
You can use bcdedit, BootICE or VisualBCD. The path of the file is `EFIESP\EFI\Microsoft\Boot\BCD`.

Intelinsidecom, 2024-01-11:
Problem was fixed, but it would need more investigation, can i use latest w10m and if not, why?

RedGreenBlue09, 2024-01-11:
How did you fix it? 

Well from W10M 15063 and newer, the Windows Boot Manager is updated so that it only loads Windows Phone and nothing else. There are some workaround such as replacing the old one, but that's kinda dirty and I haven't tested that yet.

closed 2024-01-11 by RedGreenBlue09

Intelinsidecom, 2024-01-11:
I just redid all process from original ffu to w10m 14393.x, but just with backuping. By the way, how i myself could replace win boot manager?

RedGreenBlue09, 2024-01-11:
Idk