RedGreenBlue09/WFAv7_Installer #38: Lumia 630 support?

cocus, 2024-03-01:
Hi, I have an old Lumia 630 over here, and would like to use it. I saw that Lumia-drivers doesn't have a definition for this cellphone. Does any of the other offer a good match for the drivers?
This phone can have a MSM8226 or MSM8626 (Snapdragon 400).
Thanks!

RedGreenBlue09, 2024-03-02:
Drivers for another model (RM-XXX) will not load on your phone. For this, you need to use GetLumiaBSP to extract drivers from WP, but my installer doesn't implemented that yet.

There is another installer which use GetLumiaBSP, but the link is gone so consider joining the Telegram group to get it.

RedGreenBlue09, 2025-04-10:
I just developed a new experimental installation method which supports more devices including the Lumia 630. You can try it out in this branch: https://github.com/RedGreenBlue09/WFAv7_Installer/tree/dev/generic_inf

closed 2025-04-15 by RedGreenBlue09