RedGreenBlue09/WFAv7_Installer #42: Lumia 650 Sad face

Weectowe, 2024-05-06:
After successfull installation with drivers my lumia after restart from mass storage mode gives sad face

RedGreenBlue09, 2024-05-07:
Can you give me the log?

Weectowe, 2024-05-07:
from what, your installer?

Weectowe, 2024-05-07:
Here you go, and at first, your installer didnt detect mainos partiton automatically, I had to edit Installer.bat and instead of MainOS is:%DriveLetter%: I had to change it to MainOS is: F:

#### Windows 10 for ARMv7 Installer 3.1.3 #### 
#### INSTALLATION WAS STARTED AT 20:14:03.00 #### 
========================================================= 
## Device is 7  ## 
## MainOS is F: ## 
 
## MainOS DriveLetter is F: ## 
## Disk number is 2 ## 
## DPP PN is 1 ## 
## EFIESP PN is 35 ## 
## Data PN is 33 ## 
## Dualboot is n ## 
## ChargeThreshold is 0 ## 
## DebugEnabled is n ## 
The type of the file system is NTFS.
Volume label is Data.

Stage 1: Examining basic file system structure ...
Progress: 0 of 3840 done; Stage:  0%; Total:  0%; ETA:   0:00:13    
Progress: 3840 of 3840 done; Stage: 100%; Total: 28%; ETA:   0:00:09 .  
                                                                                       
                                                                                       
  3840 file records processed.                                                        
File verification completed.
 Phase duration (File record verification): 303.21 milliseconds.
Progress: 6 of 6 done; Stage: 100%; Total: 13%; ETA:   0:00:14 .. 
                                                                                       
                                                                                       
  6 large file records processed.                                   
 Phase duration (Orphan file record recovery): 0.22 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 13%; ETA:   0:00:14 ...
                                                                                       
                                                                                       
  0 bad file records processed.                                     
 Phase duration (Bad file record checking): 0.16 milliseconds.

Stage 2: Examining file name linkage ...
Progress: 3843 of 9690 done; Stage: 39%; Total: 27%; ETA:   0:00:20    
Progress: 44 of 44 done; Stage: 100%; Total: 28%; ETA:   0:00:20 .  
                                                                                       
                                                                                       
  44 reparse records processed.                                      
Progress: 5767 of 9690 done; Stage: 59%; Total: 38%; ETA:   0:00:17 .. 
Progress: 9690 of 9690 done; Stage: 100%; Total: 44%; ETA:   0:00:01 ...
                                                                                       
                                                                                       
  9690 index entries processed.                                                       
Index verification completed.
 Phase duration (Index verification): 982.40 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 44%; ETA:   0:00:01    
                                                                                       
                                                                                       
  0 unindexed files scanned.                                        
 Phase duration (Orphan reconnection): 2.70 milliseconds.
Progress: 0 of 0 done; Stage: 99%; Total: 44%; ETA:   0:00:01 .  
                                                                                       
                                                                                       
  0 unindexed files recovered to lost and found.                    
 Phase duration (Orphan recovery to lost and found): 12.36 milliseconds.
Progress: 44 of 44 done; Stage: 100%; Total: 44%; ETA:   0:00:01 .. 
                                                                                       
                                                                                       
  44 reparse records processed.                                      
 Phase duration (Reparse point and Object ID verification): 1.29 milliseconds.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
 Phase duration (Security descriptor verification): 178.20 milliseconds.
[... 516 lines cut ...]
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
The operation completed successfully.
========================================================= 
The operation completed successfully.

Microsoft DiskPart version 10.0.22621.1

Copyright (C) Microsoft Corporation.
On computer: DESKTOP-QQDFDTE

Disk 2 is now the selected disk.

Partition 35 is now the selected partition.

DiskPart successfully set the partition ID.
        1 file(s) copied.
#### INSTALLATION COMPLETED SUCCESSFULLY ####

Weectowe, 2024-05-07:
ok so i did it again and this time windows starts but windows bluescreen occurs 
[image]
[image]

Weectowe, 2024-05-07:
ok i deleted this driver and installation went through

Weectowe, 2024-05-07:
so now i have problem with screen rotation, if it rotates automatically or using settings the screen goes black, restart doesnt seem to help, any suggestions?

Weectowe, 2024-05-07:
and can I somehow plug in usb device (keyboard, mouse, usb stick etc.)?

RedGreenBlue09, 2024-05-07:
You can only use bluetooth kbd and mouse.

Weectowe, 2024-05-07:
> so now i have problem with screen rotation, if it rotates automatically or using settings the screen goes black, restart doesnt seem to help, any suggestions?

???

RedGreenBlue09, 2024-05-07:
This is a known isssue. I don't know how to fix though.

RedGreenBlue09, 2024-05-07:
Lets keep this issue only for the blue screen and anything else on Telegram instead.

gabi21yt, 2024-05-20:
i have the same issue

RedGreenBlue09, 2024-05-20:
> i have the same issue
https://github.com/RedGreenBlue09/WFAv7_Installer/issues/42#issuecomment-2098265764

gabi21yt, 2024-05-20:
OTG ?

RedGreenBlue09, 2024-05-20:
> OTG ?

Only 950s. No other Lumia have OTG.

gabi21yt, 2024-05-20:
i cant change display rotation ?

RedGreenBlue09, 2024-05-21:
> i cant change display rotation ? and do you have some internet browser for arm firefox wont run

Please discuss that in https://github.com/RedGreenBlue09/WFAv7_Installer/discussions or telegram.

RedGreenBlue09, 2025-04-30:
A new fix for this bug has been pushed. If you have time, you can test the fix by redownloading drivers reinstalling with the latest Beta version.

catmengi, 2025-05-31:
> > OTG ?
> 
> Only 950s. No other Lumia have OTG.

Is it possible to trick windows to set usb to host mode and then use powered otg adapter?

RedGreenBlue09, 2025-05-31:
IDK 😅. I've seen a similar switch program for L950, but that phone already have OTG and the program uses a method that's only possible Snap 810.