RedGreenBlue09/WFAv7_Installer #49: command not working

WalkTheEarth, 2024-10-12:
Hi all, 

Im running tiny10 on my main laptop. everytime i have to run ` for /f %%i in ('Powershell -C "(Get-Partition | ? { $_.AccessPaths -eq ''%MainOS%\Data\'' }).PartitionNumber 2>$null"') do set "PartitionNumberData=%%i"` it doesnt work. (empty PartitionNumberData)
any help?

RedGreenBlue09, 2024-10-13:
It only work if you properly set up the Data mount point in MainOS like default. Maybe send me the screenshot of the mount point?

WalkTheEarth, 2024-10-13:
It has not been detected manually, but I have set it to E:

RedGreenBlue09, 2024-10-13:
I mean the Data folder in MainOS

WalkTheEarth, 2024-10-13:
E:/Data

RedGreenBlue09, 2024-10-13:
Give me the screenshot of it in file explorer

WalkTheEarth, 2024-10-13:
[image]

WalkTheEarth, 2024-10-13:
this is E:\

RedGreenBlue09, 2024-10-14:
I don't see EFIESP abd DPP mounted. Seems like you have an unusual MainOS layout. Try flash original FFU and try again. Also use the latest installer from Releases page, don't use development versions.

WalkTheEarth, 2024-10-14:
1. I already have flashed FFU
2. I do have stable too.

WalkTheEarth, 2024-10-26:
i was debugging bc it wasnt working

RedGreenBlue09, 2024-10-26:
Please send the full log.

WalkTheEarth, 2024-10-26:
when i find my 640, wherever i put it qwq

WalkTheEarth, 2024-10-27:
SOLVED
i had to mount all partitions first then run wfa

closed 2024-10-27 by WalkTheEarth