RedGreenBlue09/WFAv7_Installer #50: Error when finishing install

WalkTheEarth, 2024-10-29:
```

RedGreenBlue09, 2024-10-30:
1. Drivers not downloaded
2. Please send image of E:\EFIESP

RedGreenBlue09, 2024-10-30:
The mount points for DPP and EFIESP in MainOS seems to be broken.

WalkTheEarth, 2024-10-30:
I'll do that when I get home (in school

WalkTheEarth, 2024-10-30:
DPP is empty

WalkTheEarth, 2024-10-30:
https://mega.nz/folder/NqFgmBaD#N4wm_-a5n-9MrNai7_ICig

WalkTheEarth, 2024-10-30:
link to it

WalkTheEarth, 2024-10-30:
> The mount points for DPP and EFIESP in MainOS seems to be broken.
> 
> Also, try to run this command (assuming MainOS is E:)
> 
> ```batchfile
> Powershell -C "(Get-Partition | ? { $_.AccessPaths -eq 'E:\DPP\' }).PartitionNumber"
> ```

works, but dpp is still empty

RedGreenBlue09, 2024-10-30:
Try mounting DPP and EFIESP to the respective folders in MainOS.
https://learn.microsoft.com/en-us/windows-server/storage/disk-management/assign-a-mount-point-folder-path-to-a-drive#mount-a-drive-as-a-folder-with-disk-management

RedGreenBlue09, 2024-10-30:
Delete everything in the existing EFIESP folder

WalkTheEarth, 2024-10-31:
> Try mounting DPP and EFIESP to the respective folders in MainOS. https://learn.microsoft.com/en-us/windows-server/storage/disk-management/assign-a-mount-point-folder-path-to-a-drive#mount-a-drive-as-a-folder-with-disk-management

dpp just dont exist
(esp do tho)

RedGreenBlue09, 2024-10-31:
DPP is the first partition. You can use diskpart to see it.

WalkTheEarth, 2024-10-31:
ok

RedGreenBlue09, 2024-11-10:
Have you solved the issue?

WalkTheEarth, 2024-11-15:
No