RedGreenBlue09/WFAv7_Installer #51: Lumia 929(Icon) blue screen

Bulboussum, 2024-12-01:
When I attempt to boot Windows 10, I get a blue screen showing only the sad face. The phone does boot into Windows mobile. [image]
[image]

[2024-12-01.log](https://github.com/user-attachments/files/17968640/2024-12-01.log)

RedGreenBlue09, 2024-12-01:
That's impossible. Windows Phone shouldn't boot until you ran the PostInstall.cmd.

RedGreenBlue09, 2024-12-01:
Did you used WPinternals 2.9 or newer to unlock it?

RedGreenBlue09, 2024-12-01:
Also, you are supposed to use the camera button to select the boot entry.

Bulboussum, 2024-12-02:
I used version 2.9.2 of WPinternals. The blue screen is from selecting Windows 10 Arm boot entry.

sigmaman17, 2025-04-02:
I am having the same issue but mine is bootlooping and i didnt select dual boot

RedGreenBlue09, 2025-10-25:
I updated the installer to hopefully turn the sad face into something more descriptive. Also please remember that updating Windows Phone while having the bootloader unlocked is not safe. You should unlock the bootloader after updating.