RedGreenBlue09/WFAv7_Installer #53: Sad face/Not entering WP. Lumia 920

MirageNesher, 2024-12-12:
Did everything according to the instructions. When starting the phone, a menu with an OS selection appears. When choosing W10ARM, BSOD instantly appears and the phone reboots after 30 seconds. When choosing WP, the Nokia logo appears, lasts for about 20 seconds, and the phone just reboots. The downloader was unlocked by WPinternals 2.8, and Windows Phone 8.1 Update 2 was installed. No root access.
[2024-12-12@s6ip2q.log](https://github.com/user-attachments/files/18116831/2024-12-12%40s6ip2q.log)

RedGreenBlue09, 2024-12-13:
Please use WPinternals 2.9. WPinternals 2.8 won't work.

MirageNesher, 2024-12-15:
> Please use WPinternals 2.9. WPinternals 2.8 won't work.

I fully resetted the phone, restored bootloader, reinstalled original FFU for the phone with WPint 2.9, updated OS to WP8.1/2 and then unlocked bootloader again. Now the installation process interrupts on the diskpart part, with propmt "installation failed".
[2024-12-15-1.log](https://github.com/user-attachments/files/18139182/2024-12-15-1.log)

MirageNesher, 2024-12-15:
Tried on another PC, same story. I think DISKPART cannot create a Windows10 partition. Its been a lot nicer if we can read the log file, but we have what we have.

MirageNesher, 2024-12-16:
I managed to change cmd encoding and finally have an readable log
[2024-12-16.log](https://github.com/user-attachments/files/18146736/2024-12-16.log)

RedGreenBlue09, 2024-12-16:
Thank you for the logs

RedGreenBlue09, 2024-12-16:
Can you try deleting `Data\Windows10.vhdx...` and then install again? Windows is known to flag healthy partitions, which it fixes itself after a few tries.

RedGreenBlue09, 2024-12-16:
Also, can you still boot to WP after the error?

MirageNesher, 2024-12-16:
Hi, thank you for your support.
I managed to get it work by setting less disk space for W10 installation (20gb vs 22gb). In my last unsuccessful installation attemps WP is booting, im just delete the Windows 10 folder and E:/data/Windwos10.vhdx and all working like it new, so its not the problem. Now I passed thru all installation procces and it working competely fine, big thanks for your application.
Dualboot working too.
But some functions isnt work:
1. Auto rotate, phone locked in portrait mode, however in settings it marked as landscape and i cannot change it (button isnt active)
2. [image] 
Cant resize the paging file, its locked on 547mb and error appears every time when windows started up.
3. Auto brightness and brightness settings, phone locked into minimal brightness.
4. Sleep mode. Phone screen turning off for a second and then brights again.
Maybe i need to reinstall drivers to make it work? Everything other seems to be fine.

MirageNesher, 2024-12-16:
> Thank you for the logs

If anyone needs it, you can easily make the command line and logs readable using the new Windows 11 feature.
Go to Control Panel > Change date, time, and number formats > Additional > Language of programs that do not support Unicode > Change system language > Beta version: Use Unicode (UTF-8) to support languages worldwide

RedGreenBlue09, 2024-12-16:
2. You can [edit the registry](https://superuser.com/questions/1099626/offline-windows-install-how-to-view-and-change-pagefile-settings) to change the page file size. I'm not sure if that will get applied because of the error. I also don't know the cause of the error.
3. On my phone, only 100% and 0% works. Anything else result in 100%.

IDK any of the other problems.

closed 2024-12-16 by RedGreenBlue09