RedGreenBlue09/WFAv7_Installer #64: Lumia 640 XL 3G touchscreen doesnt work

daniil518xd, 2025-03-22:
Cant continue setup because touchscreen simply doesnt work in my install. Yes, i have recently changed its screen, however, after installing wp8.1 back it worked just as fine as with the old screen. OTG Is not supported by any lumia (except 950's, they got continuum dock support). Installation went without any errors. Installing LTE drivers gives in result a bsod with litterally nothing on it, just qr code wich leads to the basic bing search "windows error codes".

RedGreenBlue09, 2025-03-23:
Tru replacing System32\Drivers\SynapticsTouch.sys with the one from WP 8.1.  
You can use WPinternals to get MainOS from the FFU then use 7-Zip to extract the file.

RedGreenBlue09, 2025-03-26:
What is the status on this?

RedGreenBlue09, 2025-03-26:
The OOBE you got is the one triggered when you have force restarts. Still no touch?

daniil518xd, 2025-03-26:
Maybe. And just like i said, no touch

daniil518xd, 2025-03-26:
Replaced the old touchscreen (broken), nothing happened in windows 10 OOBE setup, but it also works in windows phone 8.1

RedGreenBlue09, 2025-03-26:
You can enable kernel debug and use WinDBG to get the touch driver log (use the downloaded touch driver instead of the one from WP). Use USB debugging mode with target `WOATARGET`.