RedGreenBlue09/WFAv7_Installer #66: Lumia 920 BSOD (Sad face)

ACPIFixedFeatureButton, 2025-04-01:
I attempted to dualboot Windows 10 Mobile (14393.2551) and Windows 10 ARM on my 920. The installation apparently succeeded but the Windows 10 ARM boot entry shows a BSOD. Even after manually applying the commands in PostInstall.cmd (which did fix W10M), it still bluescreens. It’s a RM-821 flashed to 3 Italy (It was on Vietnamese WP8 firmware) firmware if that matters.

RedGreenBlue09, 2025-04-01:
Did you use WPinternals 2.9 to unlock? If yes, send the log here

ACPIFixedFeatureButton, 2025-04-02:
[2025-04-01-2.log](https://github.com/user-attachments/files/19567148/2025-04-01-2.log) It's 2.9, yeah. Should I use an older WPI version to re-unlock the phone in the meantime?

RedGreenBlue09, 2025-04-02:
Everything seem to be ok.

You can uninstall it by deleting the VHDX file in Data partition, Windows10 folder in MainOS and EFI\Microsoft\Recovery\BCD in EFIESP. Then, run
```batch
bcdedit /store <MainOS drive letter>:\EFI\Microsoft\Boot\BCD /delete {703c511b-98f3-4630-b752-6d177cbfb89c}
```

And retry with the single boot option.

ACPIFixedFeatureButton, 2025-04-02:
Alright. Sad that I can't dualboot though.

RedGreenBlue09, 2025-04-02:
You can, but IDK why it doesn't work for you. It worked for many people with L920 before. Maybe if you really want to dualboot, use an older WP version like WP 8.1 or W10M build 10586.

closed 2025-04-02 by ACPIFixedFeatureButton

RedGreenBlue09, 2025-04-02:
@ACPIFixedFeatureButton what helped fixed this?

ACPIFixedFeatureButton, 2025-09-02:
I forgot about this but saw this in my old inbox notifs lol. It was something weird with my device, it’s just allergic to Windows 10 15035 for some reason. I tried it on another 920 and it works just fine