RedGreenBlue09/WFAv7_Installer #70: Lumia 625 driver issues

ACPIFixedFeatureButton, 2025-04-17:
The phone would be stuck on an infinite spinning wheel during windeploy. After skipping that, the phone freezes on a loading cursor. Once I disable test mode, the phone boots but with a lot of things broken. Seems this issue affects all S4 Play devices

RedGreenBlue09, 2025-04-17:
This is a duplicate of #22. You used generic INF right?

RedGreenBlue09, 2025-04-17:
This is extremely sad since I have no idea what caused that and how to fix it.

RedGreenBlue09, 2025-04-17:
Maybe this is related https://github.com/bibarub/Lumia-Drivers/pull/4

RedGreenBlue09, 2025-04-17:
@sonic011gamer suggested disabling the GPU driver from the registry. Try setting `HKLM\SYSTEM\ControlSet001\services\qcdx8930\Start` to `4`.

ACPIFixedFeatureButton, 2025-04-18:
It worked. A few things aren’t working like WLAN and Sound, touch alignment is a bit off. I noticed that the WLAN adapter device would change between code 38 and code 45 sometimes.

[image]

RedGreenBlue09, 2025-04-18:
Can you send me the installation log?

ACPIFixedFeatureButton, 2025-04-19:
[2025-04-17.log](https://github.com/user-attachments/files/19817493/2025-04-17.log)

[windbg.txt](https://github.com/user-attachments/files/19817496/windbg.txt)

The sensors show Code 3. There are a lot of unknown devices with exclamation marks. I can't find the audio device.

RedGreenBlue09, 2025-04-19:
I think the WLAN issue is related to https://github.com/bibarub/Lumia-Drivers/pull/4. The sensors doesn't work for any spec A device, needs investigation. The audio driver is installed using an extremely hacky method so it doesn't work. I will implement a more proper method.

RedGreenBlue09, 2025-04-27:
A WLAN related fix has been merged. Try reinstalling with the latest 4.0 beta. Remember to redownload drivers.

ACPIFixedFeatureButton, 2025-04-30:
Is it possible to just install the newer drivers? I don’t want to put the poor Samsung eMMC under any more stress than I need to

closed 2025-10-25 by RedGreenBlue09