RedGreenBlue09/WFAv7_Installer #71: SOC_SUBSYSTEM_FAILURE Lumia 640

TheWindows10User, 2025-05-03:
How to fix it?

[image]

RedGreenBlue09, 2025-05-03:
You are probably using the wrong drivers. What is your model (RM-XXX)?

jukfiuune, 2025-06-18:
I also have the same issue. RM-1077 for me, its the dual sim model

RedGreenBlue09, 2025-06-19:
Can you send me the log?

jukfiuune, 2025-06-19:
I'm using the latest commit from master. [Here is the log.](https://pastebin.com/cKFdhWck)

RedGreenBlue09, 2025-06-19:
Try this:

If you still have W10Arm installed, use WPinternals to dump MainOS from the ffu. Mount the bin file with OSFMount, logical mode. In the mounted partition, go to Windows\System32, copy all .mbn files to System32 in Windows 10 ARM partition (replace existing ones).

If you flashed WP, boot to Mass Storage mode, copy the .mbn files to somewhere. Install Windows 10 ARM. Before first boot, replace the mbn files.

RedGreenBlue09, 2025-09-19:
Okay from what I know from the community, replacing `Windows\system32\qcsubsys8626.sys` with the one from Windows Phone might fix the issue.