RedGreenBlue09/WFAv7_Installer #72: endless black screen on Lumia 1320

MotherofaDuck, 2025-05-03:
so after installing the windows build the system normally goes thru windeploy(takes 2 hours but eventually it will show "restarting" and if you wait it just bluescreens). After the boot manager the nokia logo shows up and then it goes to a blank screen. Sometimes it will restart by itself but it just hangs to the black screen. I'm now trying again and for now i'm at the windeploy screen(strangely the text appears for just a second then disappears)

[image]

MotherofaDuck, 2025-05-03:
[image]
windeploy part just finished now im gonna try to boot it again

MotherofaDuck, 2025-05-03:
and now its at a black screen again

MotherofaDuck, 2025-05-03:
ok i got it booting now, i had to disable the graphics driver by modifying this reg key
HKLM\SYSTEM\ControlSet001\services\QCDX\Start and set it to 4

RedGreenBlue09, 2025-05-04:
can you send me the log file? The audio driver is not included though. I'll try to add support for it later.

MotherofaDuck, 2025-05-04:
Sorry where is the log file?

MotherofaDuck, 2025-05-04:
there it is 

[2025-05-03.log](https://github.com/user-attachments/files/20027006/2025-05-03.log)

my windows install is in italian so some parts of it are written in italian, i hope that doesnt cause any problems

MotherofaDuck, 2025-05-06:
i rebooted to safe mode (in my case i had to type {default}) and when i rebooted i noticed that the adreno 305(Qualcomm Adreno 305 WDDM v1.2) was being picked up without installing anything but the "Generic Non-PnP Monitor" was showing an exclamation symbol

[image] 
trying to install any driver always popped out this error

[image]

i remember having this exact issue when i tried installing windows 8.1 on a laptop but this monitor driver wasnt compatible and would result in the laptop just booting into a black screen. Windows worked tho, i managed to activate the narrator and it was working but the laptop screen was just off. Mabye its happening here too?

MotherofaDuck, 2025-05-06:
i can't install any driver so i dont know what to do now. I'll just keep it in safe mode for now. If you need the instance path for the monitor driver that shows the exclamation mark its "DISPLAY\DEFAULT_MONITOR\1&8713BCA&0&UID0"

MotherofaDuck, 2025-05-06:
damn i tried to shut it down and it just bluescreened

RedGreenBlue09, 2025-05-06:
Uninstall the Adreno (don't tick uninstall driver software), then update.

MotherofaDuck, 2025-05-06:
alright lemme try

MotherofaDuck, 2025-05-06:
its taking an abnormally long time to uninstall the driver

MotherofaDuck, 2025-05-06:
i dont know why but every time i did something the device manager just hang up on everything, if i uninstalled the driver it wouldnt install or refresh the page and windows always crashed when trying to restart/shut down. Anyways i managed to uninstall the driver and when i clicked refresh it reappeared so i tried updating it but it got stuck so i just removed the safe mode option and tried to boot and well its stuck to the blank screen again.

RedGreenBlue09, 2025-05-07:
Don't disable safe mode. You should wait for it to complete installing before rebooting to normal mode.

MotherofaDuck, 2025-05-07:
It always gave me an error about the timeout, o never managed to install it

caesar511, 2025-05-11:
this happend sames to me

MotherofaDuck, 2025-06-20:
so sorry for the long delay, i re tried to uninstall the gpu driver in safe mode and reinstall it but i never manage to do that, every time i try to uninstall or just do anything on device manager it just hangs up loading or spews up a "The operation returned because the timeout period expired." error. I NEVER manage to uninstall the gpu driver successfully because of this and so i cant fix the problem as staying outside safe mode activates the driver when you scan for hardware changes after uninstalling it and locks up in a black screen. Is there any other way i can do something with this driver outside of device manager? mabye from mass storage mode or anything else?

closed 2025-10-25 by RedGreenBlue09