RedGreenBlue09/WFAv7_Installer #73: my nokia lumia 1020 blue screen with sad face

caesar511, 2025-05-10:
hi guys i  found a way to solve this problem is connect phone to pc open wpi and click on interup process then soft reset on phone
then it will found then that flash original ffu to back to normal windows phone 

**REASONS WHY MY PHONE BRICKED**
because i founds tiny 10 so i want to try after i downloads it was install.esd then i learn how to convert to install.wim with dism after done i do any installation no error show up but when i reboot phone it boot to blue screen cause i want more new windows 10

caesar511, 2025-05-10:
after solve i try to install it again but brick this time it boot to blue screen with out any face and it reboot automatic when i try to charge the phone it show battery was low how to fix?

or it because battery that i set? cause i set to 40 at least to boot

RedGreenBlue09, 2025-05-10:
Install with the WIM provided in README. Other versions doesn't work.

closed 2025-05-10 by RedGreenBlue09