RedGreenBlue09/WFAv7_Installer #75: Lumia 550 cannot charge in windows

catmengi, 2025-05-29:
Lumia 550 cannot charge in windows, charge simply doesnt start. Cannot set charge threshold sine `bcdedit /store "F:\EFIESP\EFI\Microsoft\BOOT\BCD" /set {globalsettings} chargethreshold "50"` gives: `Недопустимая команда установки.
Для получения справки введите в командной строке "bcdedit /?"
Параметр задан неверно.`

RedGreenBlue09, 2025-05-30:
The USB controller driver needs the Qualcomm filter driver to work. As I know, only MSM8960, MSM8974 and MSM8926 has it, so not the 550.

RedGreenBlue09, 2025-05-30:
I have no idea what is wrong with your command. Remember to run it as admin and check if the file `F:\EFIESP\EFI\Microsoft\BOOT\BCD` exists.

RedGreenBlue09, 2025-05-30:
> Is it also from windows 10 mobile?

Yes, but I need to edit the INF file so that it can install correctly.

catmengi, 2025-05-30:
> Can I have the english error?

Sorry, but my windows installation is currently ruined

catmengi, 2025-05-30:
> > Is it also from windows 10 mobile?
> 
> Yes, but I need to edit the INF file so that it can install correctly.

Do you know how mount mainOS.bin from wpinternals backup to get that driver? And need to be edited, registry, install path?

catmengi, 2025-05-30:
> It's not recommended to install drivers using Regedit. If you want to mount the file, you can use OSFMount.

No, i mean what need to be edited in .inf

RedGreenBlue09, 2025-05-30:
I will do. It's not trivial.

RedGreenBlue09, 2025-05-30:
Extract this file to `Installer folder\Drivers\Lumia550`. Then, reinstall. This driver is only installed at setup as far as I know.

catmengi, 2025-05-30:
Can i reinstall windows arm without reflashing to w10m first?

catmengi, 2025-05-30:
> You will need to recreate the Data partition, then create an empty folder in MainOS (or W10 drive) and mount the new partition to that folder. MainOS auto detect will not work and you will get some errors but they're okay.

So i need to shrink win10 part, create Data named partition of any size, and then mount it to random folder in W10 partition? and then it will work?

catmengi, 2025-05-30:
> Yes and yes.

I'm redownloading ffu and wpinternals because linux fucked up this file, that's gonna take a while

catmengi, 2025-05-30:
[cut_dism_log.txt](https://github.com/user-attachments/files/20523106/cut_dism_log.txt) there is the cut log, because full log contain dism logs from 2023, your script's log file is written in garbage symbols (because of RU lang in system?)

catmengi, 2025-05-30:
I successfully installed windows 11 on main PC, and started your installer script, it works now! im waiting it to install, then test new drivers!

catmengi, 2025-05-30:
For unknown reason my lumia wipes it self (deleting MainOS part) after bootloader unlock. (fix was found: remove SD card)

catmengi, 2025-05-30:
Charging works now!

closed 2025-05-30 by catmengi